concordia-fsae/firmware #179: Missing bmsb mcu temp in yamcan

JoshLafleur, 2025-03-28:
As noted in PR TBD the mcu temp is missing from yamcan

JoshLafleur, 2025-09-13:
BMSB calculates the current mcu temperature in `components/bms_boss/ENV.c`. We need to add the `mcuTemp` signal to yamcan in `network/definition/data/components/bmsb/bmsb_(message&signals).yaml` and then set the CAN signal in `components/bms_boss/include/CANIO_componentSpecific.h`

Nurtrantpoem19, 2025-09-18:
ok so i've added a define in canio header as well as mcuTemp in bmsb signals yaml file. Not sure what to do for bmsb message yaml since there is no equivalent in bmsw message yaml.

Nurtrantpoem19, 2025-09-18:
> > ok so i've added a define in canio header as well as mcuTemp in bmsb signals yaml file. Not sure what to do for bmsb message yaml since there is no equivalent in bmsw message yaml.
> 
> For the message you just add the signal name to some lower frequency information message. Literally just mcTump being added into some message in bmsb message. Please put up a pr for review

ok done. i think. thanks